OpenVTx/OpenVTx #7: "Error: failed erasing sectors 0 to 15" while flashing E7082VM V1.2

qdrk, 2021-04-04:
Just got my E7082VM arrived, however when I try to flash it, I got the following error. Here's the board I got [aliexpress link](https://www.aliexpress.com/item/4000938942385.html?spm=a2g0o.productlist.0.0.2db0cbbdo5GBDn&algo_pvid=beba5f4c-bc6f-4709-8714-1de034577e37&algo_expid=beba5f4c-bc6f-4709-8714-1de034577e37-1&btsid=0bb0624316175159596531482e0e71&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_).

xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000
** Programming Started **
Error: stm32x device programming failed
Error: failed erasing sectors 0 to 15
embedded:startup.tcl:521: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 586
at file "embedded:startup.tcl", line 521

qdrk, 2021-04-04:
After disabling flash read out protection using ST Link Utility, now I get a different error:

```
CURRENT: upload_protocol = stlink
Uploading .pio\build\EWRF_E7082VM\firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002a5c msp: 0x20001000
** Programming Started **
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0x080014b2 msp: 0x20000fb8
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:521: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 586
at file "embedded:startup.tcl", line 521
*** [upload] Error 1
```

The flash memory shows half of it was written, the other half was still 0xff

qdrk, 2021-04-04:
Looks like I might need to connect RST pin as well according to this thread: https://github.com/vsergeev/arm-bmw-sw/issues/1

Do folks know if there's a pad for RST on the board?

qdrk, 2021-04-04:
All right, it appears to be some bug introduced in the newer commits, when I checkout to a really old commit like 17e81c17234bd1e7486c600f97c809fc88caf262, it flashed fine.

qdrk, 2021-04-04:
Looks like it started to fail from commit f70f4b28c076c849a5be063a218c52fcc367c976

dimitroffs, 2021-04-22:
I also had same error while flashing for first time same vtx.. working with commit 96bb28ba4418e636c38062b4852d6ad23b7b9314

JyeSmith, 2021-04-22:
Sorry guys, I missed the initial issue. Try the below.

I think its because the GD32 mcu it is not 100% compatible with the ST Utility.  So you will get errors but it will still work.

[image]

[image]

closed 2021-05-10 by JyeSmith

derFrickler, 2021-05-21:
Just tried to flash my first VTX after removing the write protection and doing a full chip erase via ST-Link utility. 
End up with the same error:

Uploading .pio/build/EWRF_E7082VM/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002ba4 msp: 0x20000f80
** Programming Started **
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0x080014ee msp: 0x20000f60
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:521: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 586
at file "embedded:startup.tcl", line 521
*** [upload] Error 1

JyeSmith, 2021-05-21:
Did you tick the 3 boxes for the option byte when disabling write protection?

derFrickler, 2021-05-22:
Tried again with the boxes tickend and after that flashed via pio on windows. that worked fine. 
Will investigate if it was the boxes or a pio problem on linux.