OpenVTx/OpenVTx #12: Support EWRF E7092VM and Inav?

s2kaka, 2021-05-15:
I'm working on tiny long range 2s project and very happy when see openvtx.
(Looking for low voltage but high power and cheap vtx.)
I have some quest want to ask:

1. EWRF E7082VM with openvtx can use with inav or not?
2. Will openvtx support EWRF E7092 with already have 500mw output to higher power (600-800mw)?

Thanks for develope!

JyeSmith, 2021-05-15:
It isn't currently working with INav, but I have a fix ready for testing.  Should be working in a couple of days.

We haven't tested and other ewrf vtxs.  But they all use the GD32 and probably the same, or similar, pin out.  There's a good chance it will work, but be prepared for it to not work and require some dev work.

s2kaka, 2021-05-16:
> It isn't currently working with INav, but I have a fix ready for testing. Should be working in a couple of days.
>
> We haven't tested and other ewrf vtxs. But they all use the GD32 and probably the [...]

Thank for reply and wait for your new! 90% of my build are using inav

s2kaka, 2021-05-17:
could i config it on betaflight then move it to inav to use 500mw?

JyeSmith, 2021-05-17:
@s2kaka you can test this branch out https://github.com/OpenVTx/OpenVTx/tree/Adds-powerLevels-and-fixes-for-INav

s2kaka, 2021-05-18:
i have not have hardware yet, i ordered from alliepress so  it will come on next 10 days.

JyeSmith, 2021-05-18:
No probs.  By the time you receive your HW INav will be supported in master.

closed 2021-05-18 by JyeSmith