OpenVTx/OpenVTx #22: Use openvtx without Smart audio wire?

s2kaka, 2021-07-24:
My FC only have 2 uart for rclink and gps. 
How can i use openvtx without audio wire.
i want max power level when power it on .  
or (wiring SA to an uart, set chanel, power then unplug SA wire from uart to use) will them keep the setting?

JyeSmith, 2021-07-24:
At they moment a tlm connection is required.  The vtx is currently dumb and doesn't remember any settings.

You can use softserial.

s2kaka, 2021-07-24:
so what is it default setting now? which power level and channel?
Will it be possible in future?

JyeSmith, 2021-07-24:
Yes it's possible on the future.

Default is 4uW (pitmode) and a out of range frequency of 5000MHz.

s2kaka, 2021-07-24:
thanks! i ll wait for it, i have 3 noxe v1 that use inav, can not use softserial because it same TX2 pad that use for rc link.
and i have bough 5 open vtx

s2kaka, 2021-07-24:
Can i revert to stock FW to use 200mw without smart audio wire?

JyeSmith, 2021-09-10:
@s2kaka eeprom is merged and you can now setup with SA and it will boot back to the previous settings.

closed 2021-09-10 by JyeSmith