OpenVTx/OpenVTx #29: Understanding of powerlevels / Emuflight

iz3man, 2021-09-10:
I watched this video https://www.youtube.com/watch?v=5lHPgiON7xA to understand how OpenVTx works, and as I understand it, the FC sends ANY wanted powerlevel (defined by DB) to the VTX. 
With betaflight that's very easy to acchieve as you have "vtxtable powervalues x y z", where xyz stands for the wanted DB output, correct?
Unfortunately there is nothing like defineable powervalues in Emuflight. I can use Tramp and enable/disable pitmode, change power levels (25, 100, 200, 400, 800mW), but I have no idea what the 200mW setting really corresponds to? 

a) Is it really 200mW? 
b) And how can I set 300mW (which is the highest my VTX is able of)? 
c) What will 400 and 800 do? May they cause any damage to the VTX?

Thanks for helping sort that out :)

JyeSmith, 2021-09-10:
Use the tramp protocol and the mW sent is what you'll get.  But 400 and above just sets the vtx the max.  On the HM ovx300 range this can be up to 700mW.

[image]

iz3man, 2021-09-10:
Oh nice. I thought OVX300 was limited to 300mw (hence the name).
So to sum it up: 

.) Use TRAMP (btw: with Smartaudio I was unable to get it out of pitmode)
.) WYSIWYG: 25, 100, 200mW is exactly that
.) 400mW is the VTX's max - for the OVX300 that's ~700mW

Thanks a lot for the quick help and the great software!!

JyeSmith, 2021-09-10:
I'm currently working on targets for the HM OVX range.  At the moment the output is a bit higher than what your set.  But you aren't missing out when you set max lol

iz3man, 2021-09-10:
:) Excellent. Just ordered two more of those OVX. And btw: Love ELRS even more ;)

closed 2021-09-10 by iz3man