OpenVTx/OpenVTx #54: PlatformIO upload of OVX306 firmware, port not found? Linux Debian

j-keiter, 2022-07-27:
Not sure if I'm supposed to create a udev rule or I'm not using platform io with the stlink correctly?   Sorry if I'm asking in the wrong place.

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/gd32f130g6.html
PLATFORM: ST STM32 (15.4.1) > GigaDevice GD32F130G6U6
HARDWARE: GD32F130G6U6 72MHz, 4KB RAM, 24.00KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-spl @ 2.10300.0 (1.3.0) 
 - tool-dfuutil @ 1.11.0 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/Happymodel_OVX306/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.2% (used 2056 bytes from 4096 bytes)
Flash: [=======   ]  69.6% (used 17092 bytes from 24572 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, custom, jlink, mbed, stlink
CURRENT: upload_protocol = custom
Uploading .pio/build/Happymodel_OVX306/firmware.bin
Uploading .pio/build/Happymodel_OVX306/firmware.bin
  ** Searching flight controllers **
Traceback (most recent call last):
  ** No FC found, find all ports **
  File "/mnt/data/temp/FPV/OpenVTx-master/src/scripts/uart_upload.py", line 201, in <module>

Detected the following serial ports on this system:
    port = serials_find.get_serial_port()

  File "/mnt/data/temp/FPV/OpenVTx-master/src/scripts/serials_find.py", line 68, in get_serial_port
    raise Exception('No valid serial port detected or port already open')
Exception: No valid serial port detected or port already open
*** [upload] Error 1
================================= [FAILED] Took 0.68 seconds =================================

Environment        Status    Duration
-----------------  --------  ------------
Happymodel_OVX306  FAILED    00:00:00.681
============================ 1 failed, 0 succeeded in 00:00:00.681 ============================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'Happymodel_OVX306'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

dmesg
[16708.295279] usb 2-1: Manufacturer: STMicroelectronics
[16708.295282] usb 2-1: SerialNumber: +\x11\x12	#\x16000002
[16803.630816] usb 2-1: USB disconnect, device number 25
[16808.742254] usb 2-1: new full-speed USB device number 26 using xhci_hcd
[16808.891533] usb 2-1: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
[16808.891541] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16808.891544] usb 2-1: Product: STM32 STLink
[16808.891547] usb 2-1: Manufacturer: STMicroelectronics
[16808.891550] usb 2-1: SerialNumber: +\x11\x12	#\x16000002
[16862.761232] perf: interrupt took too long (2501 > 2500), lowering kernel.perf_event_max_sample_rate to 7975

JyeSmith, 2022-07-27:
Are you using an st link?

From the log it looks like its looking for a FC and you have selected the passthrough flashing method.  "Happymodel_OVX306" is passthought only, but you can try "EWRF_E7082VM_develop" which is the sample target but uses the st link.

j-keiter, 2022-07-28:
Thx, I did not know OVX306 is passthrough only.  Do I have to issue the cli command serialpassthrough in inav or betaflight or does vscode upload handle passthrough via smartaudio?  I got a success in vscode for uploading the ewrf bootloader to the ovx306 using the stlink.  I had to run the removeprotections before the bootoader would upload.  I have now connected back to an old F3 Evo flight controller with inav 2.52 on it.  I set uart 2 to tbs smartaudio and connected the tx2 pin on flight controller to smartaudio pin of ovx306.  I get this:  
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/gd32f130g6.html
PLATFORM: ST STM32 (15.4.1) > GigaDevice GD32F130G6U6
HARDWARE: GD32F130G6U6 72MHz, 4KB RAM, 24.00KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-spl @ 2.10300.0 (1.3.0) 
 - tool-dfuutil @ 1.11.0 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/Happymodel_OVX306/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.2% (used 2056 bytes from 4096 bytes)
Flash: [=======   ]  69.6% (used 17092 bytes from 24572 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, custom, jlink, mbed, stlink
CURRENT: upload_protocol = custom
Uploading .pio/build/Happymodel_OVX306/firmware.bin
Uploading .pio/build/Happymodel_OVX306/firmware.bin
  ** Searching flight controllers **
      > FC found from '/dev/ttyACM0'

Detected the following serial ports on this system:
  /dev/ttyACM0

=================== FIRMWARE UPLOAD ===================
  Bin file '.pio/build/Happymodel_OVX306/firmware.bin'
  Port: /dev/ttyACM0, protocol: SA
======== PASSTHROUGH INIT ========
  Trying to initialize /dev/ttyACM0 @ 4800
  Warning: 'No CLI available. Already in passthrough mode?'

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...

Attempting to reboot into bootloader...
Wait sync...[FAILED] Unable to communicate with bootloader...
Traceback (most recent call last):
  File "/mnt/data/temp/FPV/OpenVTx-master/src/scripts/uart_upload.py", line 206, in <module>
    uart_upload(port, filename, protocol)
  File "/mnt/data/temp/FPV/OpenVTx-master/src/scripts/uart_upload.py", line 121, in uart_upload
    raise Exception(msg)
Exception: [FAILED] Unable to communicate with bootloader...

*** [upload] Error 1
================================= [FAILED] Took 41.00 seconds =================================

Environment        Status    Duration
-----------------  --------  ------------
Happymodel_OVX306  FAILED    00:00:40.997
============================ 1 failed, 0 succeeded in 00:00:40.997 ============================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'Happymodel_OVX306'" terminated with exit code: 1.   

I have tried with serialpassthrough 1 115200 rxtx while connected to uart2 then disconnected inav configurator.  I have also tried without the serialpassthrough command.

JyeSmith, 2022-07-28:
You dont have to do anything beyond setting up the FC with the correct port and SA.  PFIO will then detect these settings.  Just make sure the INav Configurator is closed, or anything else that may try and grab the com port.

You can also try using openvtx.org

j-keiter, 2022-07-28:
I made sure inav configurator and the stcubeprogrammer were both closed then tried upload of ovx306 from vscode again.  Still getting the Attempting to reboot into bootloader...  error.

j-keiter, 2022-09-13:
I think setting udev rules fixed my problem.  I'm sorry I just have not had time to go back and confirm.  Here is a link incase any other linux users are having issues uploading firmware via vscode.  https://github.com/iNavFlight/inav/blob/master/docs/USB%20Flashing.md

closed 2022-09-13 by j-keiter