OpenVTx/OpenVTx #55: About my vtx's Migration issues

HanBingze123, 2022-08-03:
Hello,I made a vtx from RTC6705 and RFPA5542.
Can you tell me how to control power with RFPA5542? 
I appreciate your reply.

JyeSmith, 2022-08-03:
Its done via PWM to control the voltage bias applied to the RTC6705 output.

HanBingze123, 2022-08-03:
Thank you!
Do you know which pin of RTC6705 is connect to RFPA5542's
power control pin?
RFPA5542 has three vcc pins，Is it through them to control power?
And PFPA5542 has a "PDET" pin,Is it a output pin or a input pin?
Thanks.

JyeSmith, 2022-08-03:
I started a design but never finished, built or tested it.  This should give you some ideas.

https://oshwlab.com/jyesmith/openvtx-rfpa5542

HanBingze123, 2022-08-03:
I saw it.
I is useful for my board.
But I still have some questions...
1.Power control is in RTC6705 or RFPA5542?
2.I saw your PWM controlled voltage,but it connect with a Connector?(I guess it is a ESP32 board)
3.If power control is in RFPA5542,Which pin controls it？
Thanks

HanBingze123, 2022-08-05:
I' sorry,are you right here?

closed 2022-10-30 by JyeSmith