OpenVTx/OpenVTx #62: Difference frequencies OpenVTX & Skyzone

Michielhier, 2023-01-21:
Good day People,

I'm trying to connect my openVTX Happymodel OVX 303 on Low Race band to my Skyzone Sky02x but this doesn't work and I see that the frequencies differ.
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/wilma/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/wilma/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8
-- | -- | -- | -- | -- | -- | -- | -- | --
Open vtx | 5333 | 5373 | 5414 | 5453 | 5493 | 5533 | 5573 | 5613
Skyzone | 5362 | 5399 | 5436 | 5473 | 5510 | 5547 | 5584 | 5621

</body>

</html>

Any idea which are the correct lowband frequencies and how I can correct them?

Thank you kindly in advance,

cruwaller, 2023-01-21:
hmm, you are correct. There seems to be a bit variation in frequencies on low race band... 
Question will be which ones are the correct...? @JyeSmith have you seen any "official definition" for that band? Or should we consider to drop that whole band to avoid confusion?

JyeSmith, 2023-01-30:
Skyzone is wrong here compared to the rest of the community.  Closing this issue but I will also forward it Skyzone for their info and hopefully its and easy change with a FW update.

https://github.com/betaflight/firmware-presets

OVTX - `vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917`

TBS - `vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917`

IMRC - `vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917`

Tramp-Nano-Instruction-Manual-English-2.pdf

https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf

MultiGP
https://docs.google.com/document/d/1jWVjCnoIGdW1j_bklrbg-0D24c3x6YG5m_vmF7faG-U/edit#heading=h.vv6tqgn3t985

cruwaller, 2023-01-30:
Thanks @JyeSmith for the clarification!

Michielhier, 2023-03-07:
Thank you for reopening..
I thought your answer was final :) looking forward to the fix.

TheRealMrSteel, 2023-03-08:
I've posted a similar question like this a couple of days ago. The more I looked into this, the more I understand how easy it is to get confused.
What I didn't know till now: apparently Lowband and Low Raceband are both a thing and I'm not sure if the names and frequencies are always used in the right way.
Probably the main question is: Does OpenVTx support Low band, Low Raceband or possibly both? I think in the current form it has Lowband, but no Low Raceband.

Tried to collect some numbers to get a clear overview (hopefully):
|                    | 1    | 2    | 3    | 4    | 5    | 6    | 7    | 8    |
|--------------------|------|------|------|------|------|------|------|------|
| OpenVTX Band L| 5333 | 5373 | 5414 | 5453 | 5493 | 5533 | 5573 | 5613 |
| Lowband?** (Band L) | 5333 | 5373 | 5413 | 5453 | 5493 | 5533 | 5573 | 5613 |
| Low Raceband***     | 5362 | 5399 | 5436 | 5473 | 5510 | 5547 | 5584 | 5621 |
| Skyzone Lowband    | 5362 | 5399 | 5436 | 5473 | 5510 | 5547 | 5584 | 5621 |
| Rapidfire Band L   | 5362 | 5399 | 5436 | 5473 | 5510 | 5547 | 5584 | 5621 |

** source: https://oscarliang.com/fpv-channels/ and https://adamjnowak.com/tools/fpv-racing-5-8ghz-vtx-channel-tool/?5g8=1
*** https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf and the same tool above mentions specifically that for TBS Unify the channels are in reversed order with Low Race

To be honest, I'm not even sure anymore if there is a clear consensus on:
Band L = Lowband
Band DT = Low Raceband

But at least while looking at the Channel Tool linked above, it became clear that both Skyzone and Rapidfire support Low Raceband frequencies by default.

Not sure if this helped, or made it even more confusing (I might even be incorrect with this information, but this is my attempt gathering all the info).

At least a good start to have a clear discussion about this topic.

JyeSmith, 2023-07-24:
> Probably the main question is: Does OpenVTx support Low band, Low Raceband or possibly both? I think in the current form it has Lowband, but no Low Raceband.

If you use the TRAMP protocol you can use any frequency you like in the VTx table.  TRAMP is not confined to a set freq table, or power levels, like SA.  One of the nice things about that protocol.

closed 2023-10-04 by JyeSmith