OpenVTx/OpenVTx #66: add new hardware, necessary requeriments for work with openvtx

sl1pkn07, 2023-06-16:
Hi

exist a bunch of vtx in the market, but only a few units work with this firmware

what is the neccesary hardware requeriments for work with openvtx?

greetings

JyeSmith, 2023-06-21:
For now only the GD32F130 is supported.

sl1pkn07, 2023-06-21:
then. only the main mcu is the only thing needed for add support to openvtx?, not the RF module or something?

JyeSmith, 2023-06-26:
Correct.

The rtc6705 is standard across all analog VTx hardware.  The amps vary a little but control is all the same or very similar.

sl1pkn07, 2023-06-26:
i saw in the paltformio, have "support" to stm8 chips, but is disabled. why? i just found my Diatone Mamba Ultra 1000V have these mcu. is not worth the effort for add suport it?

greetings

cruwaller, 2023-06-26:
Latest version is totally untested (since gd32 addition). Feel free to test and report back 👍

Dmytrohdp, 2025-03-02:
@cruwaller  flashed mamba vtx with stm8. Works on 5800mhz, but does not change frequency. Ready to test and report back again if you make any changes in the code