OpenVTx/OpenVTx #71: OpenVTX without FC changes channels every boot

dbloemhard, 2023-11-24:
Running an OpenVTX EWRF E7082VM on a Silverware whoop, so no smart audio connection. It is on 0.2.0 so it has button press functionality. This works, however each time I plug a lipo in, the channel it starts up on is 1 further than saved.

If i set it to R1, the next time it boots, its on R2. Then the next time it is on R3 and so on. This is particularly painful because the current button press functionality requires me to press the button 47 times to get it back to the original channel (see previous issue logged).

Not sure if the bug is on the saving, or retrieving of the channel settings from EEPROM.

CapnBry, 2024-01-18:
I'd like to report this happens as well if the wire to the flight controller is dodgy. My SmartAudo wire came loose and my vtx would change channel maybe not every boot but maybe half the time. I tried to disable SA on the FC, then literally cut the wire when it was on my assigned channel so I could fly, but next power up it shifted again so this has to be on the OpenVTX side.

CapnBry, 2024-01-26:
@dbloemhard You may want to try editing the source to see if it resolves your issue. I made a pull request with the change that works for me.

Or flash the bin file (unzip it from the zip) [using the web flasher](https://openvtx.org/) in "local" mode
[fyujon-pullup.zip](https://github.com/OpenVTx/OpenVTx/files/14067674/fyujon-pullup.zip)